chirimen-oh/any-issues #148: ライセンスに関する検証

gurezo, 2016-10-11:
Ogaki Mini Maker Faire 2016やOpen Source Hardware Association Certificationに関連して
CHIRIMENのライセンス改変の検証をする

- [ ] 主語（MJ）を変えて良いか？
- [ ] 主語（MJ）を変えて矛盾が生じないか？

gurezo, 2016-10-11:
Open Source Hardware ライセンス
http://certificate.oshwa.org/

https://medium.com/@kotobuki/open-source-hardware-association-certification-9ba1adcc8995#.h7gxmip9g

tadfmac, 2016-11-05:
> 個人的な意見です

>Ogaki Mini Maker Faire 2016やOpen Source Hardware Association Certificationに関連して
CHIRIMENのライセンス改変の検証をする

とありますが、OMMF2016は現行ライセンスに抵触するような活動は行いませんので、本議論の論拠から外して頂いて良い認識です

kohichi000000, 2016-11-07:
多分、OMMF2016における本コミュニティ出展のレギュレーション云々ではなく、
小林先生に会える可能性が高いから（雑談できる）という位置づけではないですかね？

tadfmac, 2016-11-07:
@kohichi000000 さん
ご指摘のような意図があるにせよ（その議論の時に私も居たので承知のうえ）、本issueの論拠のようにも見える記載は誤解を招くと思いあえて記載させて頂きました。

gurezo, 2016-11-19:
11/08 のMTGで、問題ない見解でまとまったので、close

closed 2016-11-19 by gurezo