chirimen-oh/any-issues #152: I2CSlaveDevice.write16() がwrite8と同じ動作になる

tadfmac, 2016-11-02:
2箇所の実装によるもののようです。

- [ ] 1. [Polyfill](https://github.com/club-wot/WebGPIO)の[write16()](https://github.com/club-wot/WebGPIO/blob/draft-20160125/src/WebI2C/interfaces/I2CSlaveDevice/I2CSlaveDevice.js#L126)の中でwrite8()と同じ関数を呼び出している

　[関連 issue](https://github.com/club-wot/WebGPIO/issues/6)

- [ ] 2. Geckoに実装されている[I2cService.WriteWord()](https://github.com/chirimen-oh/gecko-dev/blob/b2g44_v2_5-chirimen/dom/i2c/I2cService.h#L43)の第３引数の型が`uint8_t`となっており、2byteのデータが渡せない

tadfmac, 2016-11-02:
先に 2. からPull Requestを発行させていただく予定でおります。

tadfmac, 2016-11-03:
そして、Polyfill側の修正についてもPRを発行させていただきました。
https://github.com/club-wot/WebGPIO/pull/21

※ビルド方法がわからず、@gurezo さんに多大なるご協力を頂きました。 この場を借りて御礼申し上げます。

MSakamaki, 2016-11-04:
> club-wot/WebGPIO#21

こちら確認してマージしておきました、最新に適用済みです。

tadfmac, 2016-11-04:
@MSakamaki さん、Polyfill側修正反映いただき、ありがとうございます！ 
本issueは、polyfill側とgecko側の2箇所直す必要がありますので、下記PR反映まで本issueはOpenのままとさせてください。
https://github.com/club-wot/WebGPIO/pull/21
よろしくお願いします。

MSakamaki, 2016-11-04:
@tacoyaky  あ、すいません間違えてこちらcloseしちゃってました、了解です。

closed 2016-11-04 by dadaa