chirimen-oh/any-issues #153: 電子情報通信学会の東京支部教育イベントの公募

naokisekiguchi, 2016-11-08:
東京支部教育イベントの公募
http://www.ieice.org/tokyo/koubokyouikuibento.html
にCHIRIMENイベントを提案する
締め切りは11/11

naokisekiguchi, 2016-11-15:
提案しました。
12月中に採択かどうかの通知が来る予定です。

naokisekiguchi, 2017-05-23:
本件、採択されました。
８月開催予定なので、いろいろと準備を進めていきたいです。

naokisekiguchi, 2017-05-25:
テーマ：ウェブ技術で作るオリジナルゲームコントローラー
オープンソースでいじれるウェブゲームを募集します。教えてください。

naokisekiguchi, 2017-05-25:
webgl / playcanvas 版 tank https://playcanv.as/p/y4412Fkr/

WhiteHawk-taka, 2017-05-25:
これはどうでしょうか？
https://github.com/clojurecup2014/astrocats
http://astrocats.net/

tadfmac, 2017-05-26:
> @WhiteHawk-taka さん
+1

> @naokisekiguchi さん
個人的には「CHIRIMENを持って操作」させるのは、モニターのケーブルもあるし、ちょっと無理があるんじゃないかと感じます。昨日USB接続という話も出たのですが、それに近い（でもCHIRIMENを使う意味のある話として）CHIRIMENには[Basicシールド](https://github.com/chirimen-oh/shields/tree/master/CHIRIMENBasicShield)を繋いで、外付けにI2C接続の小さなコントローラを伸ばして、繋げば 2人対戦くらいまでなら作れないかな? という気がします。

●Groveケーブル 50cm (ちょっと短いですが...許容範囲かと)
https://www.amazon.co.jp/SEEED-ACC83057O-GROVE-4%E3%83%94%E3%83%B3%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB-50cm-5%E6%9C%AC%E3%82%BB%E3%83%83%E3%83%88/dp/B00CDFY3TO

先に何をつなぐか?  ですが、Canzasiベースで基板起こしておけば、1個¥500もかからず作れる気がします。
と思いましたので、朝通勤電車の中でちょっくら作りました。
eagle-file公開してますので、使えるようであれば、つかっちゃってください。
(Groveコネクタで繋がる5cm*5cm 以内のちっちゃなコントローラです)

https://github.com/tadfmac/Canzasi
https://github.com/tadfmac/Canzasi/tree/master/boards/Cont/v0.1/eagle

※今、elecrow 5cm*5ch $4.9と超安いのでそれに合わせましたw

naokisekiguchi, 2017-05-29:
swooop https://playcanv.as/p/JtL2iqIH/
Flappybird https://playcanv.as/p/2OlkUaxF/

naokisekiguchi, 2017-05-29:
@dynamis Flappybirdとかtankとか、オープンソースでソースコード提供されていますか？ちょっと見つからなかったので。

naokisekiguchi, 2017-06-20:
flatris（テトリス）：https://github.com/skidding/flatris
bombergirl：https://github.com/MattSkala/html5-bombergirl

naokisekiguchi, 2017-06-20:
シンプルなflappy-bird cloneのfloppy-birdというのがありました。
https://github.com/nebez/floppybird
いくつかのflappy-bird cloneを試してみましたが難易度がちょうどいい感じ+難易度調整も簡単
CHIRIMEN上で動かして、GPIO入力（タクトスイッチ）で操作可能なようにできました。
ちょっと動きが重い感じがしますが。

ライセンス的な観点では、上記のソースコード自体はapacheライセンスなので問題なし。
オリジナルのflappy-birdに関しては、上記のリンクで引用されているように、作者がhtml5版オープンソースについて「自分で作ったら？」と言っているのでOKかな？
http://i.imgur.com/AcyWyqf.png

naokisekiguchi, 2017-08-08:
イベント用の電子部品の購入部材
https://docs.google.com/spreadsheets/d/1wumTdHkvrO4urNcurJhy_5uY2NA3FFh6TqhIUtNWqfM/edit?usp=sharing

こちらにリストアップしていこうと思います。仮案を入力しておきました。

naokisekiguchi, 2017-08-08:
PCセットアップと当日のチューター参加者用の調整さんを作成しました。

https://chouseisan.com/s?h=a096fb8d0b89448b8d7f454a435bc32e

naokisekiguchi, 2017-08-08:
教育コンテンツは上記のfloppy-birdに好きなI2Cセンサで操作できるようにするものを想像してます。

CHIRIMENからgroveハブを伸ばして、そこに好きなI2Cセンサを挿すイメージ。
100円ショップで買えるような部材もある程度そろえておいて、工作込みでコントローラを作る感じはいかがかな、と。（うちわに加速度センサを貼り付ける、といったような）

@tadfmac せっかく用意してもらったのに申し訳ないですが、、

naokisekiguchi, 2017-08-15:
レンタルPCと電子部品の発注完了

naokisekiguchi, 2017-08-17:
当日のメモ帳を用意しました。
https://public.etherpad-mozilla.org/p/chirimenEdu20170819

gurezo, 2017-08-17:
CHIRIMEN教育イベント
https://chirimen-oh.connpass.com/event/61533/

念の為、イベントリンク記載

gurezo, 2017-09-12:
レポートも出来たので、close

closed 2017-09-12 by gurezo