chirimen-oh/any-issues #156: Web I2C APIの仕様de

tadfmac, 2016-11-20:
現在の[Web I2C API](https://rawgit.com/browserobo/WebI2C/master/index.html)で定義されているSlave Deviceとのデータ送受信に利用可能なAPI(read8、write8など)は、[SMBus Spec 3.0]の

tadfmac, 2016-11-20:
下記のように進めるべきかと思い、CHIRIMENのissueに登録させていただきました。

1. CHIRIMENとCHIRIMENに接続することを期待するI2C Deviceなどのユースケースを考慮し追加するAPIを検討する
2. Web I2C APIへのSpecへの提案をおこなう

tadfmac, 2017-01-18:
#164 終了後に検討実施予定です。

tadfmac, 2017-02-13:
下記変更を検討中です。
ツッコミお待ちしております！

interface I2CSlaveDevice {
    readonly attribute unsigned short slaveAddress;

    Promise  read8(octet registerNumber);							// registerNumberを unsigned shortから octetに変更
    Promise  read16(octet registerNumber);							// registerNumberを unsigned shortから octetに変更
    Promise  write8(octet registerNumber, octet value);				// registerNumberを unsigned shortから octetに変更
    Promise  write16(octet registerNumber, unsigned short value);	// registerNumberを unsigned shortから octetに変更
    Promise  readBytes(octet readLength);			// 追加
    Promise  writeBytes(sequence<octet> buffer);	// 追加
    Promise  readByte();							// 追加
    Promise  writeByte(octet value);				// 追加
};

# readBytes(octet readLength)

readLengthに指定したoctet数読み込みを行う。
読み込み成功したデータを sequence で返す

# writeBytes(sequence<octet> buffer)

指定したbufferの書き込みを行う。

# readByte()

1byteだけ読み込みを行う。

# writeByte(octet value)

1byteだけ書き込みを行う。

tadfmac, 2017-02-17:
TODO:

- パケット図を追加する

gurezo, 2017-07-01:
@satakagi さん、お時間ある時にコメント下さい

tadfmac, 2017-11-01:
https://github.com/chirimen-oh/chirimen-raspi3
こちらで下記追加提案の APIを実装しています。

```
Promise  readBytes(octet readLength);              // 追加
Promise  writeBytes(sequence<octet> buffer);  // 追加
Promise  readByte();                                             // 追加
Promise  writeByte(octet value);                          // 追加
```

しかし、使ってみて、さらに下記のようなAPIも追加したくなりました。

```
Promise  readRegisterBytes(octet registerNumber, octet readLength);              // さらに追加
```

kou029w, 2019-12-19:
CHIRIMEN for Raspberry Pi同様にnode-web-i2c(npmパッケージとしてのWeb I2C APIの実装)のほうにも加えました