chirimen-oh/any-issues #157: Proposal for alliance

gurezo, 2016-11-20:
slackでもポストした、Alfonso Hernandezさんという「Astian Foundation」の代表？から
メールを頂いたので、issue を立てます。

以下本文----------------------------------------------------------------------
```
Hello

My name is Alfonso Hernandez representative of the Astian Foundation, we are an organization dedicated to the development of free software and technologies. From this we have made a fork of the Firefox OS project since the Mozilla Foundation stopped its development, this is fork called Astian OS. We want to continue with the project by applying new improvements and extending support for different platforms.

We know that Chirimen wants to orient and develop the same to apply it to its devices or other generic devices and therefore this proposal.

To join forces and resources to continue the development of a new operating system.

Thank you for your attention.

Stay tuned

-- 
AVISO LEGAL: Este correo electrónico contiene información confidencial de FUNDACIÓN ASTIAN. Si Usted no es el destinatario, le informamos que no podrá usar, retener, imprimir, copiar, distribuir o hacer público su contenido, de hacerlo podría tener consecuencias legales como las contenidas en la Ley 1273 del 5 de Enero de 2009 y todas las que le apliquen. Si ha recibido este correo por error, por favor infórmenos a contact@astian.org y bórrelo. Si usted es el destinatario, le solicitamos mantener reserva sobre el contenido, los datos o información de contacto del remitente y en general sobre la información de este documento y/o archivos adjuntos, a no ser que exista una autorización explícita.

LEGAL NOTICE: This e-mail transmission contains confidential information ASTIAN FOUNDATION. If you are not the intended recipient, you should not use, hold, print, copy, distribute or make public its content, on the contrary it could have legal repercussions as contained in Law 1273 of 5 January 2009 and all that apply .If you have received this e-mail transmission in error, please inform us at contact@astian.org or contact information of the sender and in general the information of this document or attached file, unless a written authorization exists.

Skype: Ponchale
PIN: 5EE30AE9
```
以上本文----------------------------------------------------------------------

Poncho09, 2016-11-23:
When would the next meeting be?

gurezo, 2016-11-23:
Hi! @Poncho09.
i was adjusting next month meeting day.
I'll close attendance or absence with 11/30 Wednesday PM 23:00.
https://chouseisan.com/s?h=2ae2815c06ff4cb5a5a27b2161981e09

just a moment please.

Poncho09, 2016-12-01:
Good morning.

Hi

I speak to know what the result of the meeting was and to know how they saw our proposal.

Thank.

gurezo, 2016-12-01:
Hi! @Poncho09.
Next Mnothly meeting date is 2016.12.13 19:30~ at japan timezone.
Please Waitng for Meeting after.

Poncho09, 2016-12-01:
@gurezo ok.
That is November 30 as the above advertising

gurezo, 2016-12-13:
Hi! @Poncho09 
I apologize to you.
because today, we didn't meeting.
meeting date was rescheduled at 2016.12.21 19:30~

Please Waitng for Meeting after.

Poncho09, 2016-12-13:
I forget to offer excuses, I also thought that the meeting was internal, that was determined at the meeting or deferred again?

gurezo, 2017-01-04:
Hi @Poncho09 A Happy New Year!

CHIRIMEN Open Hardware Community Opinions are this links.
https://public.etherpad-mozilla.org/p/chirimen-To-B2G

> not my Japanese understanding is terrible, I do not understand anything
i'm sorry. It can't be settled immediately, so a long time use Google translation, please.

and more, other member make #162 (Internationalization of GitHub and Slack).
thank you.

gurezo, 2017-02-15:
i had already answered, closing issue.
when i'll recieve response, reopen.

Poncho09, 2017-02-15:
@gurezo Hi how are you, we We remain interested In establishing an alliance or partnership between both organizations. Unfortunately we have not agreed on a meeting for the time difference and the dates of the meetings that are very extensive.

A hug.

Poncho09, 2017-03-13:
We being a fork of firefox OS we keep the original idea, which is to conserve and maintain gecko without implementing Android, we also develop free and open source software with multiple orientations. Among them IoT WoT, artificial intelligence among others. We want to establish a partnership with Chirimen to improve and develop a better operating system. We also want to develop a desktop operating system based on Devuan with support for webapps applications and with the power of convergence between both systems. We are very interested and with great enthusiasm to work with chirimen. Thank you