chirimen-oh/any-issues #158: CHIRIMEN周辺機器動作確認リストの作成

tadfmac, 2016-12-07:
タイトルの通りです。
下記に動作確認できたデバイスを皆で追記する形で情報収集したいです。

# USBWIFIドングル

|メーカー名|機種名|確認者|記載日|備考|
|:---|:----|:---|---:|:---|
|ReUdo|RE-UW100|@tadfmac|2016.12.7||
|PLANEX|GW-USNANO2A|@tadfmac|2016.12.7||

# USBハブ

|メーカー名|機種名|確認者|記載日|備考|
|:---|:----|:---|---:|:---|
|エレコム|U2H-PP3BBK|@tadfmac|2016.12.7|直挿しタイプのためCHIRIMENをケース等で底上げしないと物理的にCHIRIMENが傾く|

# HDMIモニター

|メーカー名|機種名|確認者|記載日|備考|
|:---|:----|:---|---:|:---|
|Diginnos（ドスパラ）|DG-NP09D|@tadfmac|2016.12.7|古いCHIRIMENで720P出力の場合、本モニタは拡大機能が無いので中央に小さく写る。1080Pの場合は見切れ等発生せず問題なし|

kohichi000000, 2016-12-08:
進め方ですが
「動作確認できた＝動いた」
ではなく、動かないなら「どう動かないか」も書いていってもらったほうがいいと思う。
そういう意味ではつないでみたものをどんどん書いてもらうのがいいと思う。
別の言い方をすると「動いた（動作確認できた）」にもいろいろなレベルがあると思うのでそれも考慮。
（今、備考に書いていただいている通り）

tadfmac, 2016-12-08:
動かなかった場合も書くということですね。
了解です。
ちょっと今から用事で(移動中)すぐできないので夜やります

gurezo, 2016-12-08:
製品のリンクも貼っておいた方が、後々楽かもしれません。

tadfmac, 2016-12-08:
@gurezo あざーす

naokisekiguchi, 2016-12-12:
一部の組み合わせで動作せず、というのはどう書いたらいいでしょう？△にしちゃいましたが。
英語化するときに悩みそう。

tadfmac, 2016-12-12:
@naokisekiguchi さん
ありがとうございます。
動作制限などの事象が発生したときの書き方は、一旦は書く人の主観で書いていただいていいと思います。今回の事例は△でいいかと。そのうちたくさん事例が集まってきたときに、良い書き方の収斂方法も見えてくると思ってますので

tadfmac, 2016-12-12:
USB電源アダプタを忘れていましたので、追加します。
こちらは、どういった環境で動いた、動かなかったという報告をお願いします。

> ommfで、 @gurezo さん、 @WhiteHawk-taka さんがハマった状況など、普通に発生しうる問題だと思いますので、起動ちゃんとできなかった（リセット繰り返す）などの報告も重要です。

ご協力宜しくお願いします。

sizuhiko, 2016-12-15:
手元にあるものはすべて入力しました

gurezo, 2016-12-15:
追記しました。

tadfmac, 2016-12-22:
皆様、ご協力ありがとうございます！
大体必要な情報が集まりましたので、一旦集約のうえ、webへ反映させていただきます。

tadfmac, 2016-12-22:
webに反映後closeします

tadfmac, 2017-01-18:
一旦Closeします。今後は wiki を更新ねがいます。
https://github.com/chirimen-oh/chirimen-oh.github.io/wiki/peripherals(ja)
https://github.com/chirimen-oh/chirimen-oh.github.io/wiki/peripherals(en)

closed 2017-01-18 by tadfmac