chirimen-oh/any-issues #159: CHIRIMENのGPIO/I2Cへのデバイス/パーツ接続動作確認リストの作成

tadfmac, 2016-12-12:
タイトルの通りです。
下記に動作確認できたデバイス/パーツ/を皆で追記する（コメントを追記するのではなく、本欄を直接編集・表に行を追加する）形で情報収集したいです。
→適当なタイミングで[このページ](https://chirimen.org/docs/ja/peripherals.html)に追記していく。

- 「製品名」欄には、製品名と、入手先のリンクを貼ってください。
- 「動作」欄は、期待動作＝「◯」、動かなかった＝「×」、一部制限あり＝「△」
- 「回路」「サンプルコード」欄には下記形式でURLを貼ってください。GPIOやI2Cにつなぐモノの多くはPartsです。繋げば動くというデバイスとは違うものが多く含まれますので、どのような回路でつないだのか？、その回路をもとに、どのようなコードで動かしたのか？がセットで情報収集できていないと意味がありません。

"[回路](URL)"
"[サンプルコード](URL)"

# GPIO

|製品名|回路|サンプルコード|動作|確認者|記載日|備考(動作状況等)|
|:---|:---|:----------|:--:|:---|---:|:---|

# I2C

|製品名|回路|サンプルコード|動作|確認者|記載日|備考(動作状況等)|
|:---|:---|:----------|:--:|:---|---:|:---|

tadfmac, 2016-12-14:
定例会で依頼させて頂きましたが、確認用のファームウエアバージョンは下記でお願いします。
・https://github.com/chirimen-oh/release/releases/tag/CMN2015-1_B2GOS-20160808

sizuhiko, 2016-12-14:
確認の実装方法としては、WebGPIOを使った場合という事にしますか？

tadfmac, 2016-12-14:
@sizuhiko さん
ありがとうございます。
はい。ご認識の通りです。

- Web GPIO APIを利用したデバイス/パーツの確認結果
- Web I2C API を利用したデバイス/パーツの確認結果

に記載を修正しました。

dynamis, 2016-12-15:
いろんなセンサーが使えるよ感がないとツライので、動くのは自明なんですけど GPIO デバイスと ADC 経由でのアナログセンサー類を一気に書き足すとかすると良いと思いますが如何でしょう？

具体的には (CHIRIMEN じゃないやつ用のからで恐縮ですが) こういった感じのセンサー類を (これよりももっと) 一気にリストに追加するイメージ:
https://github.com/mozilla-japan/gecko-embedded/wiki/Grove
なお、上記ページに記載のある Grove モジュールは私が私物で持ってるので Mozilla オフィスに来て頂くときなどにテスト頂けます。

tadfmac, 2016-12-16:
今回本issueで作成したいのは「動作確認リスト」なので、動作確認できていないパーツをリストに加えるのは、個人的には、賛成できません。
特に、Groveモジュールのうちアナログ入力のものはCHRIIMENで基本的にモジュール側が意図した接続を行うことはできません。(HackとしてAnalog入力の高電圧付近をデジタルのHighとして用いることもできる、というような用法もなくはないですが、それはAnalog出力のセンサーが接続できる、ということとは異なります)
頂いたリストには、上記のようなセンサーも多く含まれていますので、各モジュールの仕様を見ながらCHIRIMENが持つGPIOやI2Cに適合しているモジュールを確認していかないと、本issueの意味が無いように思えます。

dynamis, 2016-12-16:
勿論、未確認の状態で確認で追加するのではなく、間違いなく動くだろうものの動作をさっと確認して追加するという意図です。

アナログセンサ via ADC については直接じゃないのでちょっと反則かもしれませんが、「ADC xxxxx 経由で接続を確認したデバイス」などというかたちの表も追加して書いておくと、いろんなセンサーが使える感が出て良いと考えています。PWM サーボとかも同様です。

別 Issue にすべきだったかも知れませんが、ある程度のセンサーのリストが無いと何も繋がらない何も出来ない印象になってしまうのを何とかしたいからです。

tadfmac, 2016-12-16:
了解いたしました。貴重なご意見ありがとうございます。

tadfmac, 2016-12-16:
念のため、本日時点で [chirimen-oh/examples](https://github.com/chirimen-oh/examples)に登録されている全てのデバイスが[最新ファーム(CMN2015-1_B2GOS-20160808)](https://github.com/chirimen-oh/release/releases/tag/CMN2015-1_B2GOS-20160808)で動作確認できましたので更新しておきました。引き続き、追加デバイスについての情報更新をお願いします。

tadfmac, 2017-01-16:
こちらに作成予定です。
https://github.com/chirimen-oh/chirimen-oh.github.io/wiki/parts(ja)

tadfmac, 2017-01-18:
一旦Closeします。今後は wiki を更新ねがいます。
https://github.com/chirimen-oh/chirimen-oh.github.io/wiki/parts(ja)
https://github.com/chirimen-oh/chirimen-oh.github.io/wiki/parts(en)

closed 2017-01-18 by tadfmac