chirimen-oh/any-issues #161: CHIRIMEN タッチ ＆ トライ イベント

gurezo, 2016-12-23:
### TODO
- [ ] [CHIRIMEN タッチ ＆ トライ イベント](http://chirimen-oh.connpass.com/event/37934/)
- [ ] 日時：2017/01/28(土) 10:00-(仮)
- [ ] 場所：[モジラジャパン](https://maps.google.com/maps/api/staticmap?center=%E3%80%92106-0032+%E6%9D%B1%E4%BA%AC%E9%83%BD%E6%B8%AF%E5%8C%BA%E5%85%AD%E6%9C%AC%E6%9C%A8+7-5-6&language=ja&zoom=15&markers=%E3%80%92106-0032+%E6%9D%B1%E4%BA%AC%E9%83%BD%E6%B8%AF%E5%8C%BA%E5%85%AD%E6%9C%AC%E6%9C%A8+7-5-6&sensor=false&size=238x238)
- [ ] [物品リストの作成](#) 
- [x] タイムスケジュール作成（暫定）
  - 10:00 開場
  - 10:30 オープニング
  - 10:40 オリエンテーション（WebIDEの使い方とか？）
  - 11:00 タッチ＆トライタイム
  - 12:00 昼食（各自）
  - 13:00 タッチ＆トライタイム
  - 18:30 懇親会
  - 20:00 撤収
- [x] 人数調整
  * 見学枠  10人
  * CHIRIMEN Board [Echigo Rev.1](https://www.switch-science.com/catalog/2833/) レンタル枠  10人
  * CHIRIMEN Board [Echigo Rev.1](https://www.switch-science.com/catalog/2833/) 持ち込み枠  10人
  * チューター枠  10人
  * 合計：40人

dynamis, 2016-12-27:
note: Mozilla では現在 55 枚の在庫が有り 1 割くらいに故障や不具合がある想定です。参加人数的には 10-20 で大丈夫かと思いますが、40 枚くらいまでなら貸し出し可能かと思います。

tadfmac, 2016-12-27:
貸し出し用CHIRIMEN本体（+同梱付属品）以外に用意するものとしては、
- モニター
- マウス
- WIFIドングル
- USBハブ
- USB電源アダプター
を貸し出し用CHIRIMENボードと同数集める感じでしょうか？

gurezo, 2016-12-27:
[物品リスト](https://docs.google.com/spreadsheets/d/1vey-wZEaHpN2Tg2eQ0mR4JTbk_ghsvpgJsMUIDkSIsE/edit?usp=sharing)を作成しました。
上記機材を提供できる方、記入をお願い致します。

tadfmac, 2016-12-28:
記入しました。
あと、こういう形で一時的に集める場合、誰の所有物か？が解るようにシールを貼っていただく等併せてお願いした方が良いと思いました。返却の際の混乱避けるため

kohichi000000, 2016-12-28:
今回のタッチアンドトライは参加してくださった方にCHIRIMEN（の開発方法）を知っていただくことがメインである一方で、すでに知っている方も何名かいると思うので、
* 完全初心者エリア（スクール系）
* 自由行動エリア（チュータ対応系）
の２通りがあるといいなと思っています。で、完全初心者エリア用に教材が必要と思っていて、関口さんが専門学校向けに使ったものをベースにできればと思っています（若干修正が必要になるので、modify中）。

dynamis, 2017-01-06:
2/4 当日東京にいなくて参加できないのですが、事前に送付か預けるかして貸し出し可能な私物の数を追記しました。Mozilla オフィス分のディスプレイとかは @dadaa さん担当として書かれている分だと想定していますが、現状ある数は未確認、要再確認です。@dadaa さんも当日来れないと聞いてるので、私の私物とあわせて確認すれば良い感じでしょうか？

なお、WiFIドングルとか小さいモノにまでシール貼るのはちょっと面倒な気がするので自分で貸しだしたものを纏めて写真撮っておいて、その写真見ながら最後に仕分けして返却とかでも有りかも？

gurezo, 2017-01-06:
@dynamis さん、ありがとうございます。
了解しました！

dynamis, 2017-01-06:
あ、弊社オフィスで実施するんだから送付とかは必要ないですね。取り敢えず纏めてカウントしてどこかにまとめて置いておきますね。

tadfmac, 2017-01-16:
AIが不足しているような気がしましたので2点追加しておきました。
下記対応予定と認識していますが、認識違いがありましたらご指摘ください。
- 物品リストに関して、協力者からの機材収集・返却方法の決定 @gurezo さん
- スクール系エリア用教材の準備 @kohichi000000 さん

gurezo, 2017-02-03:
ざっくりスタッフタイムスケジュール書きました
schedule of CHIRIMEN Touch And Try Events for staff
https://gist.github.com/gurezo/f6f8efa2832fdb37e61c421de0af3fa1

gurezo, 2017-02-05:
イベントが終了したので close

closed 2017-02-05 by gurezo