chirimen-oh/any-issues #162: GitHubやSlackの国際化

WhiteHawk-taka, 2016-12-25:
台湾でCHIRIMENのことを知っている人と会い、GitHubやSlackが日本語で難しいという指摘があったので国際的に広めていくうえで文章ややり取りを一部英語でするなど他国の人が入りやすい環境を整備する必要があると思います。

gurezo, 2017-01-20:
issueのタイトルだけ英語で書くくらいから始めてみる
上記方針になったので、一旦 close

naobsd, 2017-02-05:
何も変わっていないし、新しく参加した人にも伝わらない気がするので、reopenしときます

kohichi000000, 2017-02-15:
- Community meeting (Feb. 15th) decision :
From now anyone should use English (should NOT use Japanese) at our community's  "slack" default channel "#general".
Instead, "#JP-local" channel will be created.

- 2/15コミュニティミーティング決定事項：
今後、当コミュニティのslackのデフォルトチャネルである"#general"では英語を利用していく。
代わりにJP-localチャネルを別途作る。
(JP-localのみでお話ししないように:-)

gurezo, 2017-03-31:
- Community meeting (Mar. 30th) decision :
From now anyone should use English (should NOT use Japanese) at our community's  "slack" default channel "#general".
"#JP-local" channel will be deleted.(Reason why, hardly not used.)
this issue is closing once.

- 3/30コミュニティミーティング決定事項：
今後、当コミュニティのslackのデフォルトチャネルである"#general"では英語を利用していく。
JP-localチャネルは、削除（ほとんど使われていなかったので）
このissueは、一度closeします

closed 2017-03-31 by gurezo