chirimen-oh/any-issues #163: Linux(ubuntu 14.04) でビルド・生成したimageがWindowsのAndroidTool.exeで読み込めない

tadfmac, 2017-01-16:
CHIRIMENのB2Gイメージを下記手順で作成しています。

1. [B2G OS ビルドの必要条件(Ubuntu 14.04 LTS / Debian 7)](https://developer.mozilla.org/ja/docs/Mozilla/B2G_OS/B2G_build_prerequisites#Ubuntu_14.04_LTS_Debian_7)に記載の手順でビルド

2. `./flash.sh`で分割されたimageを生成
3. [CHIRIMEN-tools の mkupdate.sh](https://github.com/chirimen-oh/CHIRIMEN-tools) で`update.img`を生成

こうして作られた`update.img`についてですが、ubuntuからそのまま下記コマンドでCHIRIMENに焼きこみできています。

> cd CHIRIMEN-tools/rockdev-rk3066
> ../Linux_Upgrade_Tool_v1.21/upgrade_tool uf ./update.img

一方で、上記手順で作成されたイメージを[AndroidTool_v2.33](https://github.com/chirimen-oh/CHIRIMEN-tools)を用いてWindowsから焼きこみを行おうとすると、下記のようにイメージロードに失敗するため、CHIRIMENへのアップロードが行えない状態です。

[image]

何か解決策をご存じの方がいらっしゃいましたらご教示いただきたく、よろしくお願いします。

satakagi, 2017-01-16:
ここにTips作ってあります。
http://svg2.mbsrv.net/chirimen/convImage.html

naobsd, 2017-02-05:
ubuntu上でのビルド時、wineでWindows用コマンドを利用してイメージを作成する、という方法があります。いずれにしても何かしら記載が必要ですので、reopenします。

naobsd, 2017-02-21:
workaround: use commands for Windows on Linux

1. install `wine` package on ubuntu
2. run `wineconsole mkupdate.bat`