chirimen-oh/any-issues #164: Web I2C API 品質改善版のコントリビュートについて

tadfmac, 2017-01-16:
現在、CHIRIMENで利用可能なWeb I2C APIについて、例えばwrite8()呼び出し時にSlaveAddressが異なる等明らかなエラー時もエラーが返却されない等の課題があります。
B2G側の実装が足りないことにより、Polyfill側も対応していない状況で、B2GとPolyfillを併せて改善を試みている最中です。

[一時的な中間成果登録場所](https://github.com/tadfmac/chirimen-expr)

これらの実デバイスを用いた評価とコード整理後にしかるべきfork元リポジトリにPull Requestを実施予定ですが、今回の修正により Web I2C APIの PolyfillとB2G側imageを同時に更新すしないと動作しなくなってしまいます。(Native側のAPI、I2CManager.write()等のパラメータを変更しているためIF互換性がなくなっている)

また、exampleに登録された各I2Cモジュール向けのサンプル実装についてですが、こちらはそのままでも動作するものの、今回の改善により、よりシンプルな記載が可能になるはずです。
このため、exampleについても修正版を作成する必要があります。

こうした事情から、いきなり現在の最新正式リリース版バイナリと置き換えるのは差し控えたいと考えています。
ある程度の期間、新しいimage/Polyfill/exampleをβ版としてリリースし、このコミュニティの方々に利用頂くことでの評価期間を経て正式版に差し替えたい考えです。

以上を踏まえ、次回コミュニティミーティングで下記についてご意見をいただけたら幸いです。

1. β版バイナリのリリース方法 ([release](https://github.com/chirimen-oh/release/releases)への登録方法。β版を分けたい)
2. polyfillへのPRとbranchの作成について @MSakamaki さんに相談させていただきたい
3. exampleへのPRとbranchの作成について

gurezo, 2017-01-20:
beta版を[CMN2015-1_B2GOS-20170120](https://github.com/chirimen-oh/release/releases/tag/CMN2015-1_B2GOS-20170120) に配置

tadfmac, 2017-02-10:
次回コミュニティミーティングで https://github.com/chirimen-oh/release/releases/tag/CMN2015-1_B2GOS-20170120 の正式版化についてご確認いただきたく、よろしくお願いします。

tadfmac, 2017-02-21:
当初のタスクからexamlesの改善が残っている状況ですが、既存のexampleでも問題はないため、exampleの改善については別issueを立てて対応することにし、本issueは一旦Closeします。

closed 2017-02-21 by tadfmac