chirimen-oh/any-issues #165: WebGPIO/WebI2C Polyfillの chirimen-ohへのforkについて

tadfmac, 2017-02-10:
現在、CHIRIMENの各種[examples](https://github.com/chirimen-oh/examples)ではWeb GPIO API、Web I2C APIを利用するために、[Polyfill]((https://github.com/club-wot))の利用を前提としています。

現在、mozI2Cのパラメータ変更を伴う修正のため、#164 でpolyfillとgecko側を合わせて修正するPull Requestを申請中ですが、今後も同様の修正が発生する場合に、chirimen-oh側に移行した方が同期が取れて良いように思います。

(近い将来、#156 の対応を実施しようと考えていますが、同様の修正が見込まれます)

本件の是非について、ご意見を頂きたく、宜しくお願いいたします。

- [ ] forkの是非 (次回コミュニティミーティングでご確認いただきたく)
- [ ] (forkする場合) forkリポジトリの作成
- [ ] (forkする場合) ドキュメントやexamples等のリンク修正

MSakamaki, 2017-02-20:
@tadfmac 私個人としては Polyfill (club-wot/WebGPIOをchirimen-oh側に移行)の引っ越し問題ないです。
（もともと正規のPolyfillができるまでのつなぎ・・・と考えていたので）

gurezo, 2017-03-01:
@gurezo 個人としても引越しをして問題なければ、良いと考えています。
今後のメンテ管理が分かり易い場所になるので。

gurezo, 2017-03-30:
fork したので、close

tadfmac, 2017-03-30:
AI全部終わってないのでre-open

naobsd, 2017-05-25:
現状ではファームウェアに同梱しているものが最新のpolyfillとの事。

どのような形で管理していくのか引き続き検討していくのだと思いますが、少なくともrepoの名前がWebGPIOというのは適切ではないと思います。

naobsd, 2017-05-29:
ファームウェア同梱のものが最新のpolyfillという話でしたが、examplesのforCMN2015-1_B2GOS-20170301ブランチにもpolyfillが含まれているように見えます。
https://github.com/chirimen-oh/examples/tree/forCMN2015-1_B2GOS-20170301
同じものでしょうか?

naokisekiguchi, 2017-05-29:
レポジトリの名前をこのタイミングで変更すべきだと思います。WebGPIOという名前だがWebI2Cのpolyfillも入っていてややこしいため。
どんな名前がよいか候補をまず挙げてみませんか。
私は
- polyfill
とかでいいのかなと思いますが、いかがでしょう

naokisekiguchi, 2017-05-29:
@naobsd examplesのforCMN2015-1_B2GOS-20170301ブランチに含まれているpolyfillは、bower コマンドの参照先とかがメンテナンスできていなかったため、暫定的な処理がされていたものという認識です。なので同じものだと思います。
（readGpioValueだけGPIOのreadに関するバグの暫定的な対処がされている。）

naobsd, 2017-05-29:
polyfill +1

tadfmac, 2017-05-29:
polyfill +1

gurezo, 2017-05-29:
polyfill +1

MSakamaki, 2017-05-29:
polyfill +1
複数入ってるのでPolyfillsでも良いかもしれない。

naokisekiguchi, 2017-06-01:
他に出てこないので
- polyfill
- polyfills

で決戦投票しましょうか

naokisekiguchi, 2017-06-01:
polyfills +1
今後も増えるかもしれない

MSakamaki, 2017-06-01:
polyfills +1
同じく今後も増えると予想してるので

gurezo, 2017-06-01:
polyfills +1

naokisekiguchi, 2017-07-04:
polyfills
で決定します。

naokisekiguchi, 2017-07-04:
レポジトリの名前を変更しました。
https://github.com/chirimen-oh/polyfills
しばらくは（同じ名前のレポジトリ作られるまでは）旧URLでもリダイレクトされるらしいです。

naokisekiguchi, 2017-07-04:
polyfill関係のリンク
・レポジトリへのリンク
https://github.com/chirimen-oh/polyfills
・polyfillのjsファイルへのリンク
https://github.com/chirimen-oh/polyfills/tree/draft-20160125/dist
以下の各ファイル
・polyfill取得コマンド（案）
mkdir dist
cd dist
curl -O  "https://raw.githubusercontent.com/chirimen-oh/polyfills/draft-20160125/dist/{webgpio,webgpio.min,webi2c,webi2c.min,worker.gpio,worker.gpio.min,worker.i2c,worker.i2c.min}.js"