chirimen-oh/any-issues #168: reorganize repositories on github

naobsd, 2017-02-16:
リポジトリ一覧で6つまでpinできるが、リポジトリが多すぎて選べないので整理したい。

- community運営系
  - https://github.com/chirimen-oh/any-issues
  - https://github.com/chirimen-oh/meeting
  - https://github.com/chirimen-oh/Support-And-FAQ
    - 機能していないので廃止?
  - "community" 的なリポジトリにまとめられないか
    - ロゴ等のコミュニティ的な素材置き場はここ?
- データ配布系
  - https://github.com/chirimen-oh/release
  - https://github.com/chirimen-oh/CHIRIMEN-tools
    - 解体する
    - 焼き焼きツールは release に
    - ボード図・写真は素材置き場に
    - ドキュメントは(まだ必要があれば)webに
- アプリ作成系
  - https://github.com/chirimen-oh/examples
  - https://github.com/chirimen-oh/cording-style-tests
  - https://github.com/chirimen-oh/WoTSignage
  - ルール・テンプレ的なものと、実際のアプリ、の2種類くらいには分ける必要がありそう
- ハード作成系
  - https://github.com/chirimen-oh/shields
  - https://github.com/chirimen-oh/Cases
  - まとめられないか?
- イベント系
  - https://github.com/chirimen-oh/MakerFaireTokyo2016
  - イベント事に増えていくのはどうなのか?
  - アプリ系とハード系のリポジトリに入るのでは?

ご意見募集中

naobsd, 2017-02-16:
releaseはpinしてよさそうな気がしたのでとりあえずreleaseだけpinしてみました

WhiteHawk-taka, 2017-02-16:
examplesもpinでいいと思います

naobsd, 2017-02-16:
ああ、releaseに何か入れるときは「CHIRIMEN Open Hardware利用規約」を考慮しないといけないですね

naobsd, 2017-02-16:
examplesもpinしました

naobsd, 2017-02-16:
あ、(ルール・テンプレ的なものを除く)アプリ系、APIの使い方のサンプルと、なにがしかの作品(アプリ・センサ・ボード等の組み合わせ)、の二種類だったりするんですかね...

- API使い方の例 examples
- 作品 WoTSignage/MakerFaireTokyo2016

みたいな分け方? (exampleも「センサ一つ(とワイヤ少々)だけの単純な作例」のような気はするのですが)

個別の作品は別に全部 pin しなくてもよいかな (example だけ pin しておけばいいかな) とも思いますが、古い作品は基本的に更新されないだろうから下の方に沈んでいく事になるだろうと思います。

naobsd, 2017-02-16:
- プログラミングガイド系
  - https://github.com/chirimen-oh/examples (pinned)
  - https://github.com/chirimen-oh/cording-style-tests
- ハード系
  - https://github.com/chirimen-oh/shields
  - https://github.com/chirimen-oh/Cases
- 作品系
  - https://github.com/chirimen-oh/MakerFaireTokyo2016
  - https://github.com/chirimen-oh/WoTSignage

こう?

tadfmac, 2017-02-16:
ギクゥ...
(OMMFのOpenSource化を私もグンマーさんもサボっているような...)

naobsd, 2017-02-18:
最初のコメントをupdateしました。

作品系は別にpinしなくてもいいのかなと思い始めましたが、せっかくの成果なので目につきやすい所にあるといいのではという気持ちはあります。(イベントレポートがwebにあればよいのかな?)

naobsd, 2017-02-18:
community運用系で普段触るものをpinするか、成果(作品系)をpinするか。(集約できるなら両方pinできる)

まず chirimen.org を見てもらって誘導できればいいといえばいい気もするが、なんとなく github.com は成果物の発表の場なのではという気もする。

tadfmac, 2017-02-18:
作品系は、(今後の追加は) リンク集の方が分散しなくていいような...

naobsd, 2017-02-18:
作品をgithubのインタフェース上で存在を主張(pin)しなくていいのなら、chirimen-ohで個別のリポジトリでもよいですし、chirimen-ohではない所でもよいと思います。

tadfmac, 2017-03-22:
ｺｿｰﾘ更新 (ommfのｿｰｽｱｹﾞﾙﾉﾜｽﾚﾃﾀ)
https://github.com/tadfmac/ommf2016