chirimen-oh/any-issues #184: Enrtry of Maker Faire Tokyo 2017

gurezo, 2017-04-21:
## issue purpose
- about Enrtry of Maker Faire Tokyo 2017 

## Close conditions
- [ ] enrty
- [ ] showspace
- [ ] about Selling of Echigo rev.1

gurezo, 2017-04-27:
entry apply done.

gurezo, 2017-05-01:
entry details
```
［申込内容]
1-01. 出展者名:
CHIRIMEN Open Hardware

1-02. 出展者名ふりがな:
CHIRIMEN Open Hardware

1-03. 出展者名の英文表記:
CHIRIMEN Open Hardware

1-04. 代表者名（姓）:
@gurezo last name

1-05. 代表者名（名）:
@gurezo first name

1-06. 代表者名（姓）ふりがな:
@gurezo last name

1-07. 代表者名（名）ふりがな:
@gurezo first name

1-08. 代表者のメールアドレス:
@gurezo mail address

1-10. 代表者の携帯電話番号:
@gurezo phone no

1-11. 代表者の住所（郵便番号）:
@gurezo zips

1-12. 代表者の住所（都道府県）:
@gurezo address

1-13. 代表者の住所（市区町村・番地・建物名・部屋番号・社名）:
@gurezo address

1-14. 出展者のプロフィール:
IoT や WoT は、様々なモノがコンピュータネットワークで繋がる社会において、
重要な存在です。
ヒトとモノがウェブを介して互いに協調しあえる環境が必要です。
それはウェブブラウザを核に作られていくのはないかと考えます。
CHIRIMEN は、このような環境（プラットホーム）をオープンにデザインし、
新たな社会で、ウェブを介したよりよいヒト・モノの関係とは何かを考え、提案していきます。

1-15. 出展区分:
Maker

1-16. 出展者タグの枚数:
5枚

2-01. 作品またはプロジェクトの名称:
CHIRIMEN Web & IoT

2-02. 作品またはプロジェクトの名称の英文表記:
CHIRIMEN Web & IoT

2-03. 出展内容の紹介:
CHIRIMEN Open Hardware でのタッチ＆トライ・イベントやハッカソンで制作された Web と Real （各センサー）が繋がった作品を展示する予定です。
また、CHIRIMEN での、IoT 開発を体験して頂く為にミニワークショップを開催します。

2-04-a. ウェブサイトのURL（1）:
https://chirimen.org/

2-04-b. ウェブサイトのURL（2）:

2-05. Twitterアカウント:
chirimen_oh

2-06-a. 出展カテゴリー（第1希望）:
IoT（Internet of Things）

2-06-b. 出展カテゴリー（第2希望）:
エレクトロニクス（電子工作）

2-07-a. 写真や動画のURL（1）:
https://drive.google.com/open?id=0B4V6iMhRJyKONGtYeWZTMU5jZFE

2-07-b. 写真や動画のURL（2）:
https://youtu.be/igPrE2pR8dw

2-07-c. 写真や動画のURL（3）:

2-08. 会場に持ち込む作品と機材:
›PC10台 CHIRIMEN Echigo.rev1 10枚 モバイル・ルーター10台、各種センサー機器一式
展示品付属機材一式
CHIRIMENを使った 作品外寸：50x50x50 ３作品

2-09. 必要な電源容量（W数）:
901〜1000W

3-01. 展示スペースの種類:
広いスペース

3-02. テーブルの数:

3-03-a. 広いスペース:
4200mm * 4200mm

3-03-b. 広いスペースのテーブルの数:
3

3-04. 椅子の数:
4

3-05. ハンズオン（ミニワークショップ）用テーブル:
希望する

3-06. ハンズオンのタイトル:
CHIRIMEN ミニワークショップ

3-07. ハンズオンの内容:
CHIRIMEN ボード Echigo rev.1を使ったタッチ ＆ トライ・イベント
CHIRIMEN ボード を用いた IoT 開発の体験ワークショップ

3-08. ハンズオンの参加料:
無料です。用意された機材で、CHIRIMEN ボードの開発を体験して頂来ます。

3-09. 暗いスペース:
希望しない

3-10. その他、スペースについて:
人の出入りがし易い場所を希望します。
電源の確保がしやすい場所を希望します。

3-11. 近くで出展したい他の出展者（最大2組まで）:

4-01. プレゼンテーション:
希望しない

4-02. プレゼンテーションのタイトル:

4-03. プレゼンテーションの内容:

4-04. ワークショップ:
希望しない

4-05. ワークショップのタイトル:

4-06. ワークショップの内容:

4-07. ワークショップの回数:
回

4-08. ワークショップの時間:
分

4-09. ワークショップの参加人数:
人

4-10. ワークショップの参加料:

4-11. ワークショップ参加者の申し込み方法:

4-12. ライブパフォーマンス:
希望しない

4-13. ライブパフォーマンスのタイトル:

4-14. ライブパフォーマンスの内容:

5-01. その他、不明な点や事務局へのご要望:

送信日時:
2017/04/27 12:22:21
```

gurezo, 2017-05-25:
ブーススペースの回答をする @gurezo

gurezo, 2017-05-26:
スペースに関して、ご協力いただきありがとうございます。
4200mm*2100mmの広さに対し、テーブルの上限は2本となりますのでご注意ください。

椅子の数、再度確認中

gurezo, 2017-05-26:
```
椅子は4脚ご希望として登録させていただいております。
```

gurezo, 2017-05-31:
振込については、 @gurezoが対応します。
```
このたびは、Maker Faire Tokyo 2017へ
出展のお申し込みをいただき、誠にありがとうございます。

本メールを持ちまして、CHIRIMEN Open Hardware 様の出展を承認いたします。

6月30日（金）までに出展のために必要な出展者タグを郵送し、
出展者マニュアルをPDFデータでお送りします。

また、出展募集要項でお知らせしたように、
最終的に持ち込む作品・機材のリスト、搬入出の方法などを、
ウェブのフォームから入力していただきます。

ハンダごて、石油・アルコール類、リチウムイオン電池など、
危険物および出展物としての食品の持ち込みには申請が必要です。

これらの申請の締切は、6月15日（木）17時となりました。
フォームおよび危険物申請については、あらためてご連絡いたします。

なお、プレゼンテーションやワークショップ、ハンズオンなど展示以外で
お申し込みいただきました企画に関しては、引き続き調整を行っております。
実施の可否につきましてはあらためてご連絡いたしますので、お待ちください。

出展料および出展者パスと机の追加料金については、下記請求金額を確認の上、
6月8日（木）までに指定の振込先まで口座振込にてお支払いください。
企業出展で指定日までにお振込が難しい方は、お手数ですが事務局までお知らせください。

・振込の際には、必ず振込名義人（振込依頼人）の最初に、以下の入金確認番号を記入してください。
・振込期限までにお振込みいただけない場合は、出展をキャンセルさせていただきますので、ご注意ください。
・請求書または領収書が必要な場合は、本メールの返信にて、6月8日（木）までに、宛名と合わせてご連絡ください。
・振込手数料はご負担をお願いします。

-----------------------------------------------------------------
ご請求金額とお支払い方法
-----------------------------------------------------------------
入金確認番号：87
振込名義人と代表者名が異なる場合、入金の確認ができません。
必ず振込名義人の欄の最初に上記番号をご入力ください。

※料金は税込価格です。

■出展区分：Maker
■出展者タグ枚数：5枚
■追加の出展者タグ枚数と料金（1000円/枚）：3枚 （3000円）
■出展スペースと料金
・テーブル出展の場合：-本 （0円）
・広いスペースの場合：2100mm * 4200mm（0円）

□合計入金金額：3000円
```

gurezo, 2017-06-01:
Maker Faire Tokyo 2017 出展内容調査フォーム提出・危険物申請について

```
Maker Faire Tokyo 2017出展者の皆様へ

このたびは、Maker Faire  Tokyo 2017にご出展いただき、ありがとうございます。

最終的な出展内容、持ち込み機材と搬入出方法について
以下「Maker Toolkit」の「出展内容調査フォーム」から申請をお願いします。

Maker Toolkit（出展者向け情報ページ）
http://makezine.jp/event/mft2017/makers/
パスワード:xxxxxxxxxx

また、下の「危険物について」にあてはまる物品を持ち込まれる方は「危険物申請書」の提出も必要です。

■出展内容調査フォーム・危険物申請書申請の締切：6/15（木）17:00
必要書類は「Maker Toolkit」よりダウンロードいただけます。
締切までに申請いただけない場合には、出展を取り消すこともありますのでご注意ください。

        ※プレゼンテーション、ワークショップ、ハンズオンに使用する持ち込み品も合わせて申請をお願いします。
　　       それぞれお申し込みいただいた方の実施可否は、順次ご連絡をいたします。

        ※出展申し込みの際に、展示に関連する食品についてご相談いただいた方には、
　        順次必要な手続きについてご案内しております。

■危険物について
出展申し込みページの「7. 出展のガイドライン」にてお知らせしたように、
 東京ビッグサイトでは「東京都火災防止条例上の禁止行為」が適用されます。
 以下にあてはまる物品を持ち込む場合は 「危険物申請書」の提出をお願いします。

出展のガイドライン（危険物や防災規則などご確認ください）
 http://makezine.jp/event/mft2017/callformakers/#guideline7

1）ハンダごてなど高温になる器具（400度を超えるもの）
ただし、ハンダごてはハンズオン・ワークショップで使用するもののみ持ち込みが可能です。
作品の修理などは「出展者専用の修理スペース」（場所は出展者マニュアルにて掲載します）に
用意したハンダごてをご使用ください。

2）危険物品
ガソリン、軽油などの引火性液体、メタノール、エタノールなどのアルコール類、火薬、可燃性ガスなど

3）リチウムイオン電池、リチウムイオンポリマー電池
パソコンやスマートフォン、携帯電話、デジタルカメラなど市販の家電製品に内蔵されているリチウムイオン電池以外の、
電子部品として販売されているリチウムイオン電池（リチウムポリマー電池、リチウムフェライト電池も含む）が対象です。

■消防査察について
危険物申請内容に基づき、8/4（金）または8/5（土）の搬入・設営時間に消防署による査察が行われます。
該当の出展者の方へは、査察の時間とともに7月にご連絡をさせていただきます。

該当の出展者（出展の代表者）は消防査察に立ち会っていただく必要がありますので、
必ず該当時間にブースにて待機していただきますよう、お願いいたします。
---------------------------------------------------------------------------------------------------

※ご不明な点がございましたら、事務局（makers@makejapan.org）までお問い合わせください。
※フォーム送信後、ご記入いただいた代表者メールアドレスに確認メールが送付されます。
※車両での搬入出は、この調査フォーム締切以後はお受けできませんのでご注意ください。
　申請がない場合は車両での搬入はできません。

出展に関する「よくある質問」もご確認ください。
http://makezine.jp/event/mft2017/faq/

お手数をおかけしますが、よろしくお願いいたします。

 ---
 Maker Faire Tokyo 事務局
 makers@makejapan.org
```

tadfmac, 2017-06-02:
I think that action items are not enough listing now, for example, it need to more detailer, same as OMMF2016.

See [OMMF 2016 Action Items](https://github.com/chirimen-oh/any-issues/issues/146)

naokisekiguchi, 2017-06-02:
We should create a template of action items for events. What do you think?

gurezo, 2017-06-12:
各位
下記のようにキックオフミーティングのお誘いが来ました。
自分は出席できないので、希望者の方に申し込みサイトの
リンクとパスワードを送ります
```
こちらのメールは、Maker Faire Tokyo 2017 出展者申込のご担当者様にお送りしております。

皆さま

このたびは Maker Faire Tokyo 2017 へお申し込みいただき、誠にありがとうございます。
6月20日（火）に、出展者、スポンサー向けのキックオフミーティングを行うこととなりました。

当日の主な内容は、Maker Faire Tokyoのスタッフによる
今年の企画のプレゼンテーションと、
出展者、スポンサーによるライトニングトーク、懇親会の予定です。

開催直前のご案内となってしまい申し訳ございませんが、
出展者同士のコミュニケーションの機会として、
また出展に関するスタッフへのご相談の場として、ご活用いただければ幸いです。

すでに、出展者の皆さんの作品製作は進んでいることと思いますが、
プロジェクトの内容をここでシェアすることで、
当日のコラボレーションにもつながるかもしれません。

イベントの詳細、お申し込みは下記のPeatixのページにてご確認ください。
皆様にお会いできることを楽しみにしています！

■イベント概要
名称：Maker Faire Tokyo 2017 キックオフミーティング
日時：2017年6月20日（火）19:00 - 21:00 （受付開始：18:30）
会場：LODGE（東京都千代田区紀尾井町1-3 東京ガーデンテラス17F）
        ※当日は東京ガーデンテラス18F LODGE内「特別イベント受付」までお越しください。入館証をお渡しいたします。
対象：Maker Faire Tokyo 2017の出展者、スポンサー、パートナー
定員：100名
会費：1,000円（ソフトドリンク、スナックをご用意いたします）
申込方法：Peatix申込フォームより
        ※各出展組から2名までお申し込みいただけます。なるべく多くの出展組が参加できるよう、ご協力をお願いします。
```

tadfmac, 2017-06-27:
テーブルクロス防炎頼んだ気がするけど一応マークついてるか確認しといてくださいな。> ぐんまさん

gurezo, 2017-06-27:
@tadfmac さん、7/1ミーティングで確認しておきます

gurezo, 2017-06-30:
出展者タグを6/30に受領しました。
```
Maker Faire Tokyo 2017 出展者のみなさま

先日は「出展内容調査」および「危険物申請書」の提出に
ご協力をいただき、ありがとうございます。

出展者タグを、出展代表者の方に順次郵送しておりますので
受け取りのうえ、ご確認をお願いいたします。

追加の出展者タグが必要になった場合には、
搬入・設営日（8月4日）、イベント当日（8月5日、6日）に、
「インフォメーションカウンター」にて、代表者の方に販売します。（現金支払のみ）

搬入出、会期中は、出展者タグがない場合、会場に入ることができません。
代表者の方は事前にグループの方へ出展者タグを配布するか、
連絡を取り合い会場の外で受け渡しを行うよう、お願いします。

＜出展者マニュアル・会場レイアウト・宅配便配置伝票について＞
7月上旬にお送りする予定です。お待たせしており、申し訳ございません。
宅配便の配送手配については、出展者マニュアルでご案内いたします。

＜車両入構証について＞
搬入・搬出に使用する車両には、それぞれ1台につき1枚「出展車両入構証」が必要です。
出展車両入構証は、車両搬入出申請をされた出展者の方へ別途送付します。
```

gurezo, 2017-07-01:
[image]

gurezo, 2017-07-07:
マニュアル確認したい方は、 @gurezo まで連絡を下さい。
```
Maker Faire Tokyo 2017 出展者のみなさま

このたびは、Maker Faire Tokyo 2017に出展いただき、誠にありがとうございます。
予定より遅くなり申し訳ございません。出展者マニュアルの準備ができました。
以下のURLからダウンロードをお願いします。

---
出展者へのお知らせページ
http://makezine.jp/event/mft2017/makers/
パスワード：
---

出展者マニュアルの内容をよくお読みになった上でイベントに
ご参加くださいますようお願いいたします。
また、グループのすべてのメンバーの方に本マニュアルを
お読みいただくよう、代表者の方からのご連絡をお願いいたします。

＜会場レイアウト、宅配便配置伝票、車両搬入経路、車両入構証について＞
準備が整いましたら、ご連絡をさせていただきます。

＜出展者タグについて＞
出展者タグを、出展代表者の方に郵送いたしました。
すでにお手元に届いているかと思いますが、万が一到着していない場合は、
下記お問い合わせ先までご連絡ください。

出展者マニュアル、その他の出展に関するご質問は、
ご遠慮なくmakers@makejapan.orgまでお問い合わせください。
引き続きよろしくお願いいたします。

---
```

gurezo, 2017-07-07:
覚書
MFT2017 事前ミーティング 調整さん

gurezo, 2017-07-13:
資料を確認したい方は、 @gurezo まで連絡を下さい。
パスワードをお伝えします。
```
Maker Faire Tokyo 2017 出展者のみなさま

このたびは、Maker Faire Tokyo 2017に出展いただき、誠にありがとうございます。
会場レイアウト、展示スペース番号対応表、宅配便配置伝票の準備ができました。

以下のURLからダウンロードをお願いします。

---
出展者へのお知らせページ
http://makezine.jp/event/mft2017/makers/
パスワード：
---

引き続きよろしくお願いいたします。

```

gurezo, 2017-07-13:
エリア：E-01-03

gurezo, 2017-07-27:
申し込みしたい方は、 @gurezo まで連絡を下さい。
パスワードをお伝えします。
```
Maker Faire Tokyo 2017 出展者のみなさま

このたびは、Maker Faire Tokyo 2017に出展いただき、
誠にありがとうございます。

8/5（土）の展示終了後に行う「出展者懇親会」のご案内です。

イベント中はなかなか交流ができない出展者、スポンサー、スタッフ同士
いろいろな参加者の方と交流を深める場としてご活用ください！

■Maker Faire Tokyo 2017 出展者懇親会概要
===========================================
日時：8/5（土）20:00-22:00（予定）受付開始は19:30
会場：東京ビッグサイト フードコート＆ビアー　Eat iT！
http://www.bigsight.jp/services/shop/restaurant05/
料金：3,800円（1名）※子供料金はありません
内容：軽食＋飲み放題（1.5h）

お申し込みURL: http://ptix.co/2tC7AEz
パスワード：
★出展者の方のみ参加可能です。一般の来場者の方はお申し込みができません。
===========================================

申し込みは先着順で、定員になり次第締め切らせていただきます。
お申し込み後のキャンセル、返金は承ることができませんのでご注意ください。

皆さんのご参加をお待ちしております！

■SNSでの情報発信
出展作品、準備の様子などを、ぜひInstagramやTwitterでシェアしてください。

以下のハッシュタグをつけて投稿いただいた内容は、
Maker Faire Tokyo 2017 サイトで紹介させていただいております。

Instagram：#mft2017_ig
Twitter：#MakerFaireTokyo2017

どうぞよろしくお願いいたします。

```

gurezo, 2017-08-02:
```
Maker Faire Tokyo 2017 出展者の皆さまへ

いよいよ8/4（金）よりMaker Faire Tokyo 2017 設営が始まります。
ご準備や様々な情報の申請に、ご協力をいただき、誠にありがとうございました。
会場入り前に、以下のご確認をお願いいたします。

・入場用の出展者タグは、お忘れにならないようお気をつけ下さい。
　追加分は、搬入時間～会期中インフォメーションカウンターにて販売をいたします。
　現金のみの対応となりますので、ご注意ください。（1枚1,000円）

・テーブル、丸椅子の出展者都合による追加はできません。
　申請内容を再度ご確認の上ご参加ください。

・消防査察の対象の方は、8/5日（土）10時15分までに
　申請内容と同様の展示のご準備を終えて、
　ご自身のスペースでお待ちいただくようお願いいたします。
　未申請の危険物の使用と持ち込みは禁止されております。

・市販品含むリチウムイオン電池の取扱（充電したまま離れない）や
　高熱になるものと燃えやすいものが接近しないなど、防火対策にご協力をお願いします。

・5.8GHz帯の無線仕様はイベント運営に支障を及ぼすため禁止とさせていただきます。

・水や液体を持ち込みになり展示される方は、
　少量であってもビニールシートなどで床への養生をお願いいたします。
　対策がされていない場合、事務局より対応をお願いすることもございます。

それでは、2日間、皆さまと楽しいイベントを開催できればと思っております。
引き続き、どうぞよろしくお願いいたします。
```

gurezo, 2017-08-05:
```
宅配搬出をお申し込みのみなさまへ
（BCCにて失礼いたします）

明日がイベント最終日となりますが、
宅配での搬出をご予定の出展者の方は、
8/6（日） イベント終了後、作品・資材・機材の梱包を行い、
18:00〜19:00の間に、収集場所までお持ちください。

＜収集場所＞
ヤマト宅急便：東7ホール-D搬入出口（入退場口からみて右手）
佐川急便：東7、東8ホール間リンクスペース外の道路にトラック待機予定

また、梱包に不備がありますと荷受けをお断りする場合がございます。
搬出のお荷物につきましては、しっかりと輸送上問題のないように梱包願います。
破損等がありましても、責任は負い兼ねますので厳重な梱包をお願い致します。

なお、ヤマト宅急便では精密機器はメーカー専用箱での発送のみお取り扱いいたします。
カウンターにお持込の際には梱包した荷物に伝票をご記入し貼り付けた上でお持込下さい。
3辺合計が160cm以内また25kg以内でご利用ください。

*常温着払いのみ。着払い伝票が必要な方は、
6日（日）14:00〜インフォメーションカウンターまでお越しください。

短時間でのご対応をお願いすることとなりますが、
何卒ご協力をお願い申し上げます。
---
Maker Faire Tokyo事務局
出展者担当（makers@makejapan.org）
株式会社オライリー・ジャパン
〒160-0002 東京都新宿区四谷坂町12番22号
VORT四谷坂町1F

```

gurezo, 2017-08-10:
```
Maker Faire Tokyo 2017 出展者の皆様へ

この度は、Maker Faire Tokyo 2017 にご出展いただき誠にありがとうございました。
皆様のご協力によって、無事イベントを終了することができました。

次回のMaker Faire Tokyoにも、ぜひご出展をご検討いただけますと幸いです。
（日程などは、makezine.jpにて発表いたします）
今後ともMaker Faire Tokyoをどうぞよろしくお願いいたします。

出展者の方へ向けてのアンケートを作成いたしました。
今後の参考のためにこちらのアンケートにご協力いただけますと幸いです。

――
アンケートURL
https://goo.gl/forms/xWwbMoirvhJUbzBp1
――

※ 事務局では忘れ物を多数お預かりしております。
落とし物など、ご自身またはお知り合いでお心当たりのある方は、
ぜひ事務局までメールにてお問い合わせください。

Maker Faire Tokyo事務局一同
出展者担当（makers@makejapan.org）
株式会社オライリー・ジャパン
〒160-0002 東京都新宿区四谷坂町12番22号
VORT四谷坂町1F
```

gurezo, 2017-08-21:
無事終了したので、close

closed 2017-08-21 by gurezo