chirimen-oh/any-issues #185: Enrty of Android Bazar Confference Spring

gurezo, 2017-04-21:
## issue purpose
- about  Entry of Android Bazar Confference Spring

## Close conditions
- [x] enrty - already entry at 2017.04.18
- [ ] showspace
- [ ] about Selling of Echigo rev.1

tadfmac, 2017-04-24:
OK. If you want, I can offer the following to the exhibition:
- Canzasi
- CheenRimen and App (like as OMMF)

gurezo, 2017-04-24:
@tadfmac thank you.
i want to need your exhibition.

gurezo, 2017-05-06:
entry detail
```
メールアドレス *
 chirimen.open.hardware@gmail.com

出展者名 *
社名、団体名、個人名をお願いします
 CHIRIMEN Open Hardware コミュニティ

ご担当者様氏名 *
 @gurezo

ご担当者様連絡先(電話番号)
 @gurezo 電話番号

出展者サイトURL1
ABCサイトで紹介させて頂きます
 https://chirimen.org/

出展者サイトURL2
アプリのURL等もあればこちらにどうぞ
 未入力

出展種別 *
 グループ・コミュニティ

出展タイトル *
 CHIRIMEN & Web & IoT

出展内容 *
 CHIRIMENが織りなす、Web と Real の世界

出展人数 *
当日ブース担当をされる方の人数
 2～3名

配布物 *
チラシやノベルティの配布有無について教えてください
 あり

[配布物ありの場合]　配布物の内容
配布物の内容について教えてください
 チラシ、ステッカー

持ち込み機材 *
当日持ち込まれる機材と数量をご回答ください
 未定

その他
ご質問などをどうぞ
 展示品の情報の期限はいつまでですか？
```

gurezo, 2017-05-19:
### テーブル情報
```
会場スペースの都合上、ご希望の広さ4200mm*4200mmのご用意が難しい状況です。
CHIRIMEN Open Hardware様には4200mm*2100mmのご用意ということで、
準備を進めさせて頂ければと思います。
```

tadfmac, 2017-05-19:
? 4200mm*4200mm のスペースを依頼したの ?

gurezo, 2017-05-22:
@tadfmac Maker Faire Tokyoの方でした。失礼しました。

gurezo, 2017-05-25:
5/28入場開始時点 スタッフ
@gurezo
@kazygumi 

以降、任意合流

gurezo, 2017-05-29:
荷物も無事に届き（返り）イベントが終了したので、close

tadfmac, 2017-05-30:
Please share photos or movies !

gurezo, 2017-05-30:
@gurezo had taken a pictures.
follwing this urls
https://drive.google.com/open?id=0BxUDIy95pWBqZWxQakExQjVhd2M

gurezo, 2017-06-14:
@tadfmac tell me why this issue close conditions

tadfmac, 2017-06-15:
Thanks OK.

closed 2017-06-15 by tadfmac