chirimen-oh/any-issues #187: IEEE Tokyo Young Professionals hackathon

gurezo, 2017-04-26:
## issue purpose
- purpose somthing

## Close conditions
- [ ] something....
- [ ] something....

sizuhiko, 2017-04-26:
調整さん
https://chouseisan.com/s?h=3a7ef91d029e426784075aa425e73959

TakehiroSato, 2017-04-28:
先日はコミュニティミーティングに参加させていただき、誠にありがとうございました。
本イベントの開催にご協力をいただき、厚く御礼申し上げます。

現在Tokyo YP側でも、イベントの各日程候補につきましてメンバーの予定を調整しております。
5/2(火)頃を目処に双方の予定を付き合わせ、日程を決定できればと考えております。
何卒よろしくお願いいたします。

gurezo, 2017-05-05:
7/8(土)-9(日)に開催予定＠Slackより

gurezo, 2017-05-05:
スタッフ向け事前講習 調整さん
https://chouseisan.com/s?h=5828b6a1b33f41f0b13082d93956f911

gurezo, 2017-05-08:
### スタッフ向け事前講習日
- [x] 6/3(土) 13:30〜	
- [ ] TODO:講習内容

naobsd, 2017-05-25:
6/3事前講習を午前からやる？

gurezo, 2017-05-25:
(1) GROVE I2C タッチセンサ
(2) GROVE I2C 三軸加速度センサ ADXL345搭載 v1.3
(3) GROVE I2C デジタル光センサ
(4) 超音波センサ SRF02
(5) ADT7410使用 温度センサモジュール
(6) 赤色LED

5/25 MJ搬入完了 @naokisekiguchi

gurezo, 2017-05-25:
TODO:講習内容 gist @gurezo

gurezo, 2017-05-25:
フライヤー
http://sites.ieee.org/sb-keio/files/2017/05/tokyoYP_hack2017_v1_1.pdf

gurezo, 2017-05-27:
事前講習タイムテーブルドラフト
https://gist.github.com/gurezo/f327dce809151ba7c5920f735616458a

gurezo, 2017-05-29:
```
皆様、6/3(土)の事前講習会の参加者ですが、合計で【8名】となりました。
（すべてIEEE Tokyo YPのメンバーです）
GitHubに作成していただいたドラフトのページを参考に、環境設定を進めさせます。

今回講習を受けさせていただくYPメンバーが、7/8-9の本番で、各チームに入ってサポートができるようにしたいと考えております。
何卒よろしくお願いいたします。
```
slack より転機@5/29 21:39

gurezo, 2017-05-31:
```
合計で【9名】
```
slack より転機 5/31 15:32

gurezo, 2017-05-31:
### 6/3(土) タイムテーブル

|time | contents | by　| TODO |
|:----|:----|:----|:----|
|10:00 - | 開場 | [@kudodo](https://github.com/kudodo) , <br>[@gurezo](https://github.com/gurezo) | 入室 |
|10:15 - 10:30 | 開催前準備 |  [@kudodo](https://github.com/kudodo) , <br>[@gurezo](https://github.com/gurezo), anybody | 機材セッティング |
|10:30 - 11:00 | 受付開始 | TBD |
|11:00 - 11:10 | 始めの挨拶・会場説明 <br>(wifiなど) | [@gurezo](https://github.com/gurezo) |
|11:10 - 12:00 | 開発環境準備 | 全員 |
|12:00 - 13:00 | 昼食 | 各自適宜 |
|13:00 - 13:30 | Hello Worldまで | [@gurezo](https://github.com/gurezo) | |
|13:30 - 15:00 | [example LEDBlink](https://github.com/chirimen-oh/examples/tree/forCMN2015-1_B2GOS-20170301/LEDblink) | [@gurezo](https://github.com/gurezo) | |
|15:00 - 15:15 | 15分休憩 | | |
|15:15 - 18:00 | (1)~(5)いずれかのexample | 各自 | | |
|18:00 - 19:00 | 後片付け・整理 | 全員 | | |
|19:00 | 撤収 | 全員 | | |

### 対象センサー

```
(1) GROVE I2C タッチセンサ
(2) GROVE I2C 三軸加速度センサ ADXL345搭載 v1.3
(3) GROVE I2C デジタル光センサ
(4) 超音波センサ SRF02
(5) ADT7410使用 温度センサモジュール
```

### 用意するもの

* ノートPC
  - Firefoxブラウザ(最新Ver.)をインストールしておいてください。
  - https://www.mozilla.org/ja/firefox/new/
  - 2017.5.27 時点の最新は Ver. 53.0.3

### 対象OS

Firefoxブラウザ(最新Ver.)をインストールできるOS

- Windows 7 以降
- Mac OS X v10.6 Snow Leopard 以降
- Linux (Ubuntu 12.04 以降推奨)

### TODO
- [x] CHIRIMEN ボード数量確認 5/25
- [ ] 最新ファームウェア焼き込み（20枚？）
- [ ] CHIRIMEN ボード動作確認
- [ ] touch & try 機材数量チェック
- [ ] カリキュラムチェック

### CHIRIMENへのドングルの接続
* CHIRIMENへのドングルの接続方法はこちらをご参照ください。
https://chirimen.org/docs/ja/basic_startup.html

### CHIRIMEN起動、そして、Wi-Fi設定
* CHIRIMENを起動します。詳しくは、こちら
https://chirimen.org/docs/ja/basic_startup.html#tocAnchor-1-1-3

* 起動後、Settingアプリ(画面上の歯車アイコン）を開き、Wi-Fi を選択。
* 画面上部のWi-Fi をタップして ON に切り替え、<br>利用可能なネットワークをタップしてネットワークに接続。
* ネットワークが保護されている場合は、パスワードを入力する必要があります。<br>詳しくは、ネットワーク管理者に聞いてください。
* （インターネットに）正しくつながっているかどうかは、CHIRIMEN UI上のブラウザ等で<br>適当なサイトにアクセスするなどしてご確認ください。

### イベント用教科書
- [CHIRIMEN Hello world](http://qiita.com/kotakagi/items/3a4f7865ab784cd88cdd)
- [CHIRIMEN I2C sensor ](http://qiita.com/kotakagi/items/56bba849295cf4c81d39)
- [CHIRIMENによるWoTサイネージの作り方(part1)](https://github.com/naokisekiguchi/WoTsignageText/blob/master/chirimenWotSignage1.md)
- [CHIRIMENによるWoTサイネージの作り方(part2)](https://github.com/naokisekiguchi/WoTsignageText/blob/master/chirimenWotSignage2.md)
- [CHIRIMENによるWoTサイネージの作り方(part3)](https://github.com/naokisekiguchi/WoTsignageText/blob/master/chirimenWotSignage3.md)
- [CHIRIMENによるWoTサイネージの作り方(part4)](https://github.com/naokisekiguchi/WoTsignageText/blob/master/chirimenWotSignage4.md)
- [CHIRIMENでI2Cのセンサーを使用する（task.jsバージョン）](https://github.com/naokisekiguchi/WoTsignageText/blob/master/chirimenI2cSensors.md)
- [CHIRIMENでサーボモータを使用する（task.jsバージョン）](https://github.com/naokisekiguchi/WoTsignageText/blob/master/chirimenServo.md)
[- CHIRIMENにI2Cデバイスが繋がんない時に確認すること](http://qiita.com/tadfmac/items/8c3e79ed3d95c31ee72d)

gurezo, 2017-06-01:
【緩募】
6/8(土)イベントの準備で、CHIRIMENボード（最低9台）にrelease 20170301 版イメージを
焼き込み／書き込みしてくださる方募集しています。
@gurezo が今週平日にやる予定で居ましたが、現在体調を崩して対応が出来ないためです。

sizuhiko, 2017-06-01:
6/3ですよね？

gurezo, 2017-06-01:
6/3が正解です＞＜。

WhiteHawk-taka, 2017-06-01:
前のハッカソンで一応20170301が焼き済みなはずではないでしょうか

gurezo, 2017-06-03:
事前講習 done.
当日の覚書はこちら
https://public.etherpad-mozilla.org/p/chirimen-20170603

gurezo, 2017-06-14:
イベント当日のスタッフ・チューター募集の調整さん
https://chouseisan.com/s?h=4b1e0aef01974246b95f4e6ff49c92d9

gurezo, 2017-06-21:
イベント当日ですが、慶應義塾大学矢上キャンパスに直接お越しいただけますと幸いです。
今回、大学側の都合により、初日のみ会場が別の部屋になりました。
    《7/8(土)》16-A棟 厚生棟3階 大会議室
    《7/9(日)》14棟 創想館 セミナールーム3（14-203）

gurezo, 2017-06-27:
初日午前中の講義資料
https://github.com/gurezo/chirimen-doc/tree/master/ja
[book.pdf](https://github.com/chirimen-oh/any-issues/files/1105014/book.pdf)

pdfの体裁が崩れているのを調整中です。

gurezo, 2017-07-03:
ネット配布用リンク
https://www.dropbox.com/sh/gkrukylxzs4xscl/AAC7NX7QN-0vOC8BwN-eUhkya?dl=0

gurezo, 2017-07-03:
イベントスタッフ

|名前|7/8|7/9|
|:----|:----:|:----:|
|@gurezo|O|O|
|@WhiteHawk-taka|△|△|
|@kazygumi|O|X|
|@sizuhiko|O|O|

※△：AM10:00頃入予定
急遽変更の場合は、連絡の上
こちらを編集する。

gurezo, 2017-07-08:
当日etherpad
https://etherpad.net/p/IEEE_Tokyo

gurezo, 2017-07-12:
イベントが終了したので、close

closed 2017-07-12 by gurezo