chirimen-oh/any-issues #188: Implement WebGPIO polyfill for node backend

dynamis, 2017-05-25:
## Overview Description (Goal of this issue)
We want WebGPIO polyfill for node backend to make WebGPIO testable/usable on other devices.
This make API consideration easy and fast. Also, many other devices including production embedded devices can share same code and ecosystem.

* [CHIRIMEN] - [current WebGPIO polyfill] - [any applications]
* [node.js] - [gpio module] - [new WebGPIO polyfill] - [any applications]

## Action Itemc (Requirements to close this issue)
- [ ] select node version (6.x or later?)
- [ ] select node module for GPIO
  - https://www.npmjs.com/package/gpio ?
- [ ] implement polyfill

naokisekiguchi, 2017-08-03:
I implemented a prototype of web gpio polyfill for node backend.
https://github.com/naokisekiguchi/nodegpio

dynamis, 2017-09-30:
create repo with new implementation: #195

dynamis, 2017-10-20:
DONE by tadfmac: https://github.com/chirimen-oh/chirimen-raspi3

closed 2017-10-20 by dynamis