chirimen-oh/any-issues #192: WebIDE cann't able to use

gurezo, 2017-09-12:
## Overview Description (Goal of this issue)
Mozila had announced to change that add-on to WebExtension.
Because of its influence, these operations cann't be performed.
- Firefox ver55.0.0  cann't able to display project tree and Edit area.
- Firefox ver55.0.3  cann't able to installing WebApps

## relation information
https://bugzilla.mozilla.org/show_bug.cgi?id=1314811
https://bugzilla.mozilla.org/show_bug.cgi?id=1391641

## Action Items (Requirements to close this issue)
- [ ] We Will be need to choice any item at next meeting.
  - [ ] Will you continue to use B2G OS in the future?
  - [ ] Will you continue to use WebIDE in the future?

## Steps to Reproduce (only for bugs):
1. 
2. 
3. 

## Actual Results (only for bugs):

## Expected Results (only for bugs):

gurezo, 2017-09-12:
if you want to use WebIDE, you'll be install Firefox ver54 ESR.

dynamis, 2017-09-30:
I suggest
* to introduce Firefox ESR 52 (with no auto version up config)
  * http://ftp.mozilla.org/pub/firefox/releases/52.4.0esr/
* not to continue to commit for b2g (or successor)

dynamis, 2017-09-30:
Community agreed to do so on the meeting on 2017.09.30
We'll need to update documents to introduce FxESR 52 (will file another issue). -> close

closed 2017-09-30 by dynamis