chirimen-oh/any-issues #195: Create new repo for polyfills with node backend

dynamis, 2017-09-30:
## Overview Description (Goal of this issue)
prepare repos for new polyfills with node backend

## Action Items (Requirements to close this issue)
- [ ] rename existing polyfills to polyfills-echigo or polyfills-b2g
- [ ] create new polyfills-node or similar name repo
- [ ] todo list items...

tadfmac, 2017-10-07:
I created the repo fo the CHIRIMEN on Raspi3 environment is followings.
https://github.com/chirimen-oh/chirimen-raspi3

I'll be push it maybe on next weekend, it will be includes the following items.
- the installation document to construct CHIRIMEN environment to raspi3 from some package files.
- latest package files.  
- latest package files includes following items
  - the code examples with some devices, and schematic images to easy routing with raspi3.
  - WebGPIO/WebI2C polyfills (extended version)
  - Bridge Server for Raspi3 with npm instration.
  - Wallpaper
  - bookmarks for online resouces (tutorials, online examples, etc.)

So, I'm sorry, I don't schedule to make divided repo now.

tadfmac, 2017-10-17:
I published all the files that make up to the environment  of `CHIRIMEN for Raspberry Pi 3` to the following repo.

https://github.com/chirimen-oh/chirimen-raspi3

The method of setting up the environment is described in the following document.

https://github.com/chirimen-oh/chirimen-raspi3/blob/master/env/setup-ja.md

closed 2017-10-23 by dynamis