chirimen-oh/any-issues #196: Prepare server to distribute images/vms

dynamis, 2017-09-30:
## Overview Description (Goal of this issue)
We'll start distribute OS images with CHIRIMEN software (browsers + polyfills + node bridge server + node server). We need some server to distribute images for many boards, vms etc.

note: GitHub releases have 2GB file size limit:

https://help.github.com/articles/about-releases/
> Limitations on binary files : We don't limit the total size of your binary release files, nor the bandwidth used to deliver them. However, each individual file must be under 2 GB in size.

## Action Items (Requirements to close this issue)
- [ ] define sub domains
- [ ] prepare server
- [ ] share server access account

kudodo, 2017-10-02:
Who is controlling chirimen.org domain?

I can set up virtual host on 203.104.96.245, however to use chirimen.org domain, I need to contact the owner. Maybe Akatsuka san?

I would propose, https://download.chirimen.org/ for download server. Does it sound good to everyone?

gurezo, 2017-10-02:
> Who is controlling chirimen.org domain?
may be, @dadaa

dynamis, 2017-10-23:
latest build made by @tadfmac 
https://drive.google.com/file/d/0B4k_ph7hQTADcVVlYzR6a0lqX3M/view

We'll not update images so frequently. Maybe we should update this under download.chirimen.org manually now. Can you do it and share the download url? @kudodo