chirimen-oh/any-issues #197: Prepare CHIRIMEN distribution image for RPi3 (based on Raspbian)

dynamis, 2017-09-30:
## Overview Description (Goal of this issue)
Create CHIRIMEN sd boot image file for RPi3 based on Raspbian Stretch with desktop
https://www.raspberrypi.org/downloads/raspbian/

Raspbian don't have trademark and need not too be sensitive for it's branding (boot animation etc).
Some packages may not re-distribution (not sure) and safe to remove them:
http://karaage.hatenadiary.jp/entry/2017/06/20/073000
* Java
* Wolfram
* minecraft pi

## Action Items (Requirements to close this issue)
- [ ] add chirimen specific files
- [ ] remove unnecessary packages including license un-clear ones
- [ ] create sd image file

kohichi000000, 2017-09-30:
upload img as follows:
https://drive.google.com/open?id=0B4k_ph7hQTADVy1lOVZ0UlNwdHc

gurezo, 2017-10-01:
@kohichi000000 this links is problem.
Sometimes the download fails, and more It is 1 KB when unpacking.
OS is Windows10 Insider Preview

Strangely, masOS Sierra had succussed.
What 's happened???

gurezo, 2017-10-01:
I had relocated zip and img files to dropbox.
foloowing this url.
https://www.dropbox.com/sh/bxdud4hidy3xbnv/AABI-EKkBZevAbxaq_t79M-ja?dl=0

tadfmac, 2017-10-10:
release note for previous (Uploaded by @kohichi000000 ) OS image : (I'm sorry, in japanese)

Version Name : 2017/09/27-2

Installation : 
- このimageはRaspberry Pi 3でのみ動作検証を行なっており、本VersionはRaspberry Pi 3以外のボードはサポートしておりません。
- 8GByte以上のmicroSDカード(Class 10以上) へ書き込んでください。
- このimageは圧縮されています。初回起動後 `sudo raspi-config` から、`7 Advanced Options`→`A1 Expand Filesystem` を必ず実施してください。実施後、再起動してください。

trouble shooting:
- ~/Desktop/gc/GPIO などのexamples実行時、'wss://localhost:33330/' へ接続できないエラーが発生する場合があります。このエラーに遭遇した場合には、一度ブラウザ画面から`https://localhost:33330` へアクセス後、セキュリティエラー画面から例外許可を行なってください。
- 長時間examplesを連続実行していると熱暴走のような症状が発生することがあります。ヒートシンクを装着してください

known issues:
- Chromium browserにチュートリアルがまだ含まれておりません
- パフォーマンスのチューニングが不完全です

WhiteHawk-taka, 2017-10-12:
release note for previous (Uploaded by @kohichi000000 ) OS image : (I'm sorry, in English)

Version Name : 2017/09/27-2

Installation :

- This image has supported only RaspberryPi 3, So this version has not supported other board.
- Please write image to microSD card(more than Class 10) that more than 8GByte.
- This image have compressed. After initial activation, you must do `sudo raspi-config` and `7 Advanced Options` → `A1 Expand Filesystem`. After that, need do reboot. 
- When executing examples such as ~/Desktop/gc/GPIO, errors that can not connect to 'wss://localhost:33330/' may occur.

trouble shooting:

- If you encounter this error, access the `https://localhost:33330` from the browser, please exception permission from the security error screen.
- Symptoms such as thermal runaway may occur if examples are continuously executed for a long time. Please attach a heat sink.

known issues:

- As yet not include tutorial.
- Performance tuning is not complete.