chirimen-oh/any-issues #200: PolyFIll Touch and Try Events

gurezo, 2017-10-01:
## Overview Description (Goal of this issue)
#189 , #188 , #197 PolyFills Touch and Try, Feedback Events

## Action Items (Requirements to close this issue)
- [ ] schedule setting.
- [ ] event contents setting.
- [ ] entry setting.

## Steps to Reproduce (only for bugs):
1. 
2. 
3. 

## Actual Results (only for bugs):

## Expected Results (only for bugs):

tadfmac, 2017-10-17:
Tutorials are here : 
Hello World
https://qiita.com/tadfmac/items/82817476615fdc7394b3

チュートリアル 1. GPIO編
https://qiita.com/tadfmac/items/ebd01cfe46e30de70f3d

チュートリアル 2. I2C　基本編（ADT7410温度センサー）
https://qiita.com/tadfmac/items/36d5467f79b6fd3114fb

チュートリアル 3. I2C　応用編（その他のセンサー）
https://qiita.com/tadfmac/items/b17d8c6a35b31c495a36

チュートリアル 4. 最終回
https://qiita.com/tadfmac/items/d627f8d2fec3c5f8711b

tadfmac, 2017-10-17:
Temporary image file using for this event is here:
https://drive.google.com/open?id=0B4k_ph7hQTADcVVlYzR6a0lqX3M

Please see the following installation document as:
https://gist.github.com/tadfmac/527b31a463df0c9de8c30a598872344d

tadfmac, 2017-10-22:
Thank you for participating in this event.
The event is over, so this thread will close.

closed 2017-10-22 by tadfmac