chirimen-oh/any-issues #201: Sumit to use "Powered by Raspberry Pi" mark

dynamis, 2017-10-12:
## Overview Description (Goal of this issue)
We need to submit to this form to use "Powered by Raspberry Pi" mark within "CHIRIMNE on/for Raspberry Pi":
https://www.raspberrypi.org/trademark-rules/powered-raspberry-pi/

## Action Items (Requirements to close this issue)
- [ ] submit!
- [ ] check reply.

dynamis, 2017-10-12:
@kohichi000000 is asking about logo usage including this (set him as assignees)

kohichi000000, 2017-10-13:
I have submitted!

kohichi000000, 2017-10-17:
I have got the reply.
We need NOT "Powered by Raspberry Pi logo" BUT "link to the Raspberry Pi Foundation". 
(We cannot use logo for digital or online products.)
Raspberry Pi Foundation's Reply is as follows:

> Thank you for your interest in our “Powered by Raspberry Pi” program.
> We have reviewed your application but on this occasion will not approve this for use of the "Powered by Raspberry Pi" logo.
> The "Powered by Raspberry Pi" is intended for use on industrial or retail products that have a Raspberry Pi built into them.
> ·         For digital and online products a link to the Raspberry Pi Foundation website is sufficient. You can find the guidelines for using our logo here: https://www.raspberrypi.org/trademark-rules/
> You can find out more about the "Powered by Raspberry Pi" program here: https://www.raspberrypi.org/blog/powered-by-raspberry-pi/
>

tadfmac, 2017-10-22:
I already removed `Powered by Raspberry Pi l` mark from wallpaper at release 2017/10/17.

closed 2017-10-22 by tadfmac