chirimen-oh/any-issues #202: 2018年版チラシの作成

kazygumi, 2018-07-15:
## Overview Description
Maker Faire Tokyo 2018への出展に向けてチラシの刷新と印刷が必要となったため、内容をここで議論させてください。

### 以前のチラシ
[CHIRIMEN2016-A5チラシ.pdf](https://github.com/chirimen-oh/any-issues/files/2195452/CHIRIMEN2016-A5.pdf)

### 以前のissues
#42 

## Action Items
- [ ] 内容の決定
- [ ] デザイン反映（7/22 梶井）
- [ ] 印刷入稿（7/25 梶井 toラクスル）
- [ ] 会場搬入（8/4 梶井）

kazygumi, 2018-07-15:
以前の内容をテキストにしました。
こちらを元に修正いただけると助かります。

### 表
- CHIRIMENロゴ
- Hello, Real World
- サンプルコード
```
<!doctype html>
<html lang="en" dir="ltr">
<head>
<title>CHIRIMEN example - Led</title>
<script type="text/javascript" src="gpio.js" />
<script type="text/javascript">
'use strict';
window.addEventListener('load', function (){
// WebGPIO LED Blink
navigator.requestGPIOAccess().then(
function(gpioAccess) {
return gpioAccess;
}).then(gpio=>{
var port = gpio.ports.get(198);
var v = 0;
return port.export("out").then(()=>{
setInterval(function(){
v = v ? 0 : 1;
port.write(v);
},1000);
});
});
}, false);
</script>
</head>
<body>
</body>
</html>
```
- HTML5ロゴ
- Designed by CHIRIMEN Open Hardware

### 裏
- ボードのイメージ
- Make Real Things with Browser Technologies
    - CHIRIMEN Board is designed for Web developers to MAKE real things that are connected to the Web via Browser technologies.
- Easy Way to Make
    - GPIO/I2C WebAPIs (to control devices) are introduced into Gecko engine on CHIRIMEN.
Web developers need NOT change their own ordinary development style.
- Open and Neutral
    - Developers can get the firmware code, board design, manufacturing process, API spec, and all necessary sources from the following URL.
This community honors developer's will and work on the standeardization.
- SPECS
    - CPU
        - RK3066 (ARM® Cortex®-A9)
    - GPU
        - Mali™-400 (quad core)
    - Memory
        - DDR3 1GB
    - Storage
        - NAND Flash 8GB, microSD slot
    - Interface
        - microHDMI, microUSB (OTG),USB, GPIO, I2C, etc.
- CONTACT
    - https://chirimen.org
    - https://www.facebook.com/groups/chirimen/
    - https://twitter.com/chirimen_oh
- Designed by CHIRIMEN Open Hardware

WhiteHawk-taka, 2018-07-15:
表は大きな変更は要らないかと思いますが、CHIRIMENのボードイメージをどうするのかという問題はありますね。
コードはこんな感じになると思います。
[https://github.com/chirimen-oh/chirimen-raspi3/tree/master/gc/gpio/LEDblink](https://github.com/chirimen-oh/chirimen-raspi3/tree/master/gc/gpio/LEDblink)
```
<!doctype html>
<html>
<head>
<title>CHIRIMEN example - Led</title>
<script src="polyfill.js"></script>
<script>
'use strict';
window.addEventListener('load', function (){
// WebGPIO LED Blink
navigator.requestGPIOAccess().then(gpioAccess=>{
var port = gpioAccess.ports.get(26);
var v = 0;
return port.export("out").then(()=>{
setInterval(function(){
v = v ? 0 : 1;
port.write(v);
},1000);
});
});
}, false);
</script>
</head>
<body>
</body>
</html>
```

裏面は大きな項目は変えず、中身だけ修正でいいのではないかと思います。
スペックの部分は動作確認したセンサー類とかボードとかを表示ですかね？

kazygumi, 2018-07-15:
ご意見ありがとうございます！
下記のように修正してみようと思いますがいかがでしょうか。

### 表
- CHIRIMENロゴ
- Hello, Real World
- HTML5のロゴといろいろなセンサーのイメージ _★ボードのイメージから変更します_
- サンプルコードとLEDのイメージ _★頂いたコードに変更しました_
```
<!doctype html>
<html>
<head>
<title>CHIRIMEN example - Led</title>
<script src="polyfill.js"></script>
<script>
'use strict';
window.addEventListener('load', function (){
// WebGPIO LED Blink
navigator.requestGPIOAccess().then(gpioAccess=>{
var port = gpioAccess.ports.get(26);
var v = 0;
return port.export("out").then(()=>{
setInterval(function(){
v = v ? 0 : 1;
port.write(v);
},1000);
});
});
}, false);
</script>
</head>
<body>
</body>
</html>
```
- Designed by CHIRIMEN Open Hardware

### 裏
- いろいろなセンサーのイメージ _★ボードのイメージから変更します_
- Make Real Things with Browser Technologies
    - CHIRIMEN is designed for Web developers to MAKE real things that are connected to the Web via Browser technologies. _★「CHIRIMEN Board」から「CHIRIMEN」に変更しました_
- Easy Way to Make
    - GPIO/I2C WebAPIs (to control devices) are introduced into Gecko engine on CHIRIMEN.
Web developers need NOT change their own ordinary development style.
- Open and Neutral
    - Developers can get the firmware code, board design, manufacturing process, API spec, and all necessary sources from the following URL.
This community honors developer's will and work on the standeardization.
- Powerd by _★センサー類は更新頻度が高いと思われるためボードのみ表記しました_
    - Raspberry Pi 3 Model B+
    - Raspberry Pi 3 Model B
    - CHIRIMEN Board Echigo Rev.1
- CONTACT
    - https://chirimen.org
    - https://www.facebook.com/groups/chirimen/
    - https://twitter.com/chirimen_oh
    - https://github.com/chirimen-oh _★githubのURLを追記しました_
- Designed by CHIRIMEN Open Hardware

WhiteHawk-taka, 2018-07-16:
- Easy Way to Make の1行目
~GPIO/I2C WebAPIs (to control devices) are introduced into Gecko engine on CHIRIMEN.~
GPIO/I2C WebAPIs (to control devices) are introduced into polyfill by CHIRIMEN team.
今はFirefox OSではなく推奨動作はChromiumになっているので

同様に
- Open and Neutral の1行目
~Developers can get the firmware code, board design, manufacturing process, API spec, and all necessary sources from the following URL.~
Developers can get manufacturing process, API spec, and all necessary sources from the following URL.
でしょうか。

あと、 Echigoってまだ動くんでしょうか…？ > @tadfmac

tadfmac, 2018-07-16:
> あと、 Echigoってまだ動くんでしょうか…？ > @tadfmac
んと、CHIRIMENの従来のソフトやファームはEchigoで動きます。そっちはいじってないので。ただしメンテされてないかな？しばらく。
CHIRIMEN for Raspberry Pi3は別リポジトリに派生させてexamples含めて構成しなおしているので、こちらのコードはECHIGOへはバックポートしないとそのままはうごかないです。

kazygumi, 2018-07-23:
ご連絡遅くなりました。
頂いた内容でチラシデータを修正しましたので確認をお願いいたします。
今週中にはFIXしたいと考えています。

[CHIRIMEN2018-A5チラシ.pdf](https://github.com/chirimen-oh/any-issues/files/2219385/CHIRIMEN2018-A5.pdf)

WhiteHawk-taka, 2018-07-23:
見れる人は確認お願いします。 > @tadfmac @kohichi000000 @satakagi @hiyohiyo @naobsd @naobsd

hiyohiyo, 2018-07-23:
無茶苦茶かっこいい!! 私は良いと思います。

kohichi000000, 2018-07-24:
細かくてすいませんが、
Powerd  -->  Powered
くらいです～。

tadfmac, 2018-07-24:
(B+の動作確認間に合うのか心配....くらいで、それ以外は完璧!!!)

kazygumi, 2018-07-29:
@kohichi000000 
下記修正を加えましたが、こちらでいかがでしょうか。
・Powerd --> Powered に修正
・You can find the latest infomation(https://github.com/chirimen-oh) を追記

[CHIRIMEN2018-A5チラシ.pdf](https://github.com/chirimen-oh/any-issues/files/2238530/CHIRIMEN2018-A5.pdf)

kazygumi, 2018-07-31:
確認ありがとうございます！
こちらでFIXし、印刷に回します

kazygumi, 2018-07-31:
前回同様2,000部印刷致します。（7,776円税込）

kazygumi, 2018-08-10:
完了しましたのでCLOSEします

closed 2018-08-10 by kazygumi