chirimen-oh/any-issues #204: CHIRIMENの開発環境にmakecodeを導入してみてはどうでしょうか

satakagi, 2019-06-19:
https://github.com/chirimen-oh/chirimen-raspi3/issues/56
や、
https://github.com/chirimen-oh/chirimen-raspi3/issues/20
にも関係する話になりそうですが、より低年齢の層やmicro:bitとの開発環境の共通化も考慮して、[makecode]( https://makecode.com/about)でCHIRIMENのコード開発が可能な環境を検討してみるというのはどうでしょう。

以下がその背景です

- VSCodeのエンジンである[Monaco](https://github.com/Microsoft/monaco-editor)がMakecodeのスクリプトエディタモード時のエディタ
- いろいろ癖はあるものの、WebAppsのネイティブの開発言語Javascriptと近しい関係のTypeScriptがmicro:bitの開発言語
- 上記の言語をベースとしてブロック型のビジュアルプログラミング環境化した[Blocky](https://developers.google.com/blockly/)なので、ScratchよりWeb開発環境に近しい
- 近々 CHIRIMEN with MicroBitをリリースしようとしているので、その標準開発環境であるMakecodeと親和性があることは好ましい
- ブラウザ上で動作する統合開発環境という意味でjsbinなどとコンセプト類似性がある

dynamis, 2019-06-20:
開発環境・教育環境としてはすごく良いですよね。欠点や悩みどころもいくつかあって

- TypeScript そのままブラウザで実行できないので一工夫が必要
- いまのものはページ開いてすぐに編集しながら実機で実行するタイプではない
- 既存のチュートリアルその他はすべて JavaScript ベースなので TypeScript に合わせる必要

makecode はとても良いので出来ないことはないがそれなりの手間も掛かるのが悩ましいところだなと思ってたりします。OSS なので適当に CHIRIMEN 向けに調整 (リンク開くだけで即実行するよう整えるとか) を加えたサーバを運用してしまうなども選択肢だが...などと漠然と思っています。

satakagi, 2019-06-21:
JavaScriptを直接書くmakecodeなんてのも作れないのかなぁとも思いながら、一方でTypeScriptに教材全体を統一するというのも、可能性としてはあるかもと思ってみました。
JavaScriptとTypeScriptの関係をもっとよく理解しないと＜自分