chirimen-oh/any-issues #206: 技術書典８

gurezo, 2019-12-29:
## Overview Description (Goal of this issue)
技術書典８に出展、書籍・物販を行う。
売上は、コミニティ活動費に充てる

## Action Items (Requirements to close this issue)
- [ ] 書籍作成
  - [ ] 構成
    - [ ] コミュニティの活動ついて
    - [ ] コミュニティの歴史
    - [ ] チュートリアル
    - [ ] 作品集(TBD)
    - [ ] 表紙・挿絵(イラスト)
- [ ] 物販：[rpi3GpioTester](https://github.com/chirimen-oh/accessories/tree/master/forRaspberryPi3/rpi3GpioTester) 組み立てキット
    - [ ] ボード発注
    - [ ] 必要部品発注
    - [ ] 梱包
- [ ] 物販：TBD
- [ ] 展示スタッフ募集

## Steps to Reproduce (only for bugs):
1. 上記プランのコミュニティ承認
1. 費用・売上に関して
1. スケジュール作成

sizuhiko, 2020-01-14:
作品集（私のBlog記事）

- [ぎたー](http://blog.open.tokyo.jp/2019/08/12/maker-faire-tokyo-2019.html)
- [ぴあの](http://blog.open.tokyo.jp/2018/08/13/maker-faire-tokyo-2018.html)

gurezo, 2020-01-14:
リポジトリ作成する

gurezo, 2020-01-15:
@kazygumi 
お仕事は、お忙しいですか？
表紙・挿絵などをお願いしたいと考えていますが可能でしょうか？